afuerhoff/ioBroker.solarviewdatareader #1: Think about to fix the issues found by adapter checker

ioBrokerChecker, 2019-09-29:
I am an automatic service that looks for possible errors in ioBroker and creates an issue for it. The link below leads directly to the test:

https://adapter-check.iobroker.in/?q=https://raw.githubusercontent.com/afuerhoff/ioBroker.solarviewdatareader

- [ ] [E201] Bluefox was not found in the collaborators on NPM!. Please execute in adapter directory: "npm owner add bluefox iobroker.solarviewdatareader"

I have also found warnings that may be fixed if possible.

- [ ] [W400] Cannot find "solarviewdatareader" in latest repository

Thanks,
your automatic adapter checker.

P.S.: There is a community in Github, which supports the maintenance and further development of adapters. There you will find many experienced developers who are always ready to assist anyone. New developers are always welcome there. For more informations visit: https://github.com/iobroker-community-adapters/info

afuerhoff, 2020-03-17:
All issues fixed

closed 2020-03-17 by afuerhoff

ioBrokerChecker, 2021-05-04:
Thanks, that all bugs have been fixed.