afuerhoff/ioBroker.solarviewdatareader #10: Eigenverbrauchsabfrage

vw-arneth, 2020-04-08:
Hallo, 
ist es möglich noch einen Eigenverbrauch mit in den Adapter mit aufzunehmen? Ich lese über einen S0 Zähler an Solarview aus. dies steht dann unter dem Reiter EV auf der Homepage...

afuerhoff, 2020-04-20:
Schaue ich bei Gelegenheit mal. Kann aber noch etwas dauern.

afuerhoff, 2020-05-02:
Hallo, die Eigenverbrauchsabfrage habe ich auch implementiert. Bitte Version von Github laden und testen.

vw-arneth, 2020-05-03:
Hallo, leider zieht er werte aber falsche..... sind nicht übereinstimment mit webseite....
daily 13,3 KW
Tagesverbrauch: 6,6 kw

afuerhoff, 2020-05-03:
Ich kann selber nicht testen, da ich keinen Eigenverbrauchszähler habe. Ich schau nochmal in den Code , ob ich dort einen Fehler gemacht habe.
Kannst du mir im Loglevel debug mal die Daten zukommen lassen und auch die Daten in den Objekten.

vw-arneth, 2020-05-04:
Für mich sieht es so aus als on es überall die doppelten Zahlen sind.

Tagesverbrauch doppelt.
Monat  doppelt
Momentan Watt passt
Total doppelt
Monat doppelt

solarviewdatareader.0	2020-05-04 23:26:01.185	debug	(18549) system.adapter.admin.0: logging true
solarviewdatareader.0	2020-05-04 23:25:54.922	debug	(18549) connection closed
solarviewdatareader.0	2020-05-04 23:25:54.917	debug	(18549) 10*: {10,04,05,2020,23,25,0012.2,00056,001780,00015554,00206,000,000.0,000,000.0,000,000.0,000,000.0,00},(
solarviewdatareader.0	2020-05-04 23:25:51.917	debug	(18549) connection closed
solarviewdatareader.0	2020-05-04 23:25:51.912	debug	(18549) 22*: {22,04,05,2020,23,25,0011.5,00048,001952,00039387,00000,000,000.0,000,000.0,000,000.0,000,000.0,00},1
solarviewdatareader.0	2020-05-04 23:24:54.922	debug	(18549) connection closed
solarviewdatareader.0	2020-05-04 23:24:54.915	debug	(18549) 10*: {10,04,05,2020,23,24,0012.2,00056,001780,00015554,00205,000,000.0,000,000.0,000,000.0,000,000.0,00},&
solarviewdatareader.0	2020-05-04 23:24:51.918	debug	(18549) connection closed
solarviewdatareader.0	2020-05-04 23:24:51.912	debug	(18549) 22*: {22,04,05,2020,23,24,0011.5,00048,001952,00039387,00000,000,000.0,000,000.0,000,000.0,000,000.0,00},0
solarviewdatareader.0	2020-05-04 23:23:54.919	debug	(18549) connection closed
solarviewdatareader.0	2020-05-04 23:23:54.913	debug	(18549) 10*: {10,04,05,2020,23,23,0012.2,00056,001780,00015554,00196,000,000.0,000,000.0,000,000.0,000,000.0,00},.
solarviewdatareader.0	2020-05-04 23:23:51.919	debug	(18549) connection closed
solarviewdatareader.0	2020-05-04 23:23:51.913	debug	(18549) 22*: {22,04,05,2020,23,23,0011.5,00048,001952,00039387,00000,000,000.0,000,000.0,000,000.0,000,000.0,00},/
solarviewdatareader.0	2020-05-04 23:22:54.917	debug	(18549) connection closed
solarviewdatareader.0	2020-05-04 23:22:54.911	debug	(18549) 10*: {10,04,05,2020,23,22,0012.2,00056,001780,00015554,00181,000,000.0,000,000.0,000,000.0,000,000.0,00},'
solarviewdatareader.0	2020-05-04 23:22:51.934	debug	(18549) connection closed
solarviewdatareader.0	2020-05-04 23:22:51.925	debug	(18549) 22*: {22,04,05,2020,23,22,0011.5,00048,001952,00039387,00000,000,000.0,000,000.0,000,000.0,000,000.0,00},.
solarviewdatareader.0	2020-05-04 23:22:48.899	debug	(18549) self consumption meter 1 enabled
solarviewdatareader.0	2020-05-04 23:22:48.880	debug	(18549) photovoltaic inverter 4 enabled
solarviewdatareader.0	2020-05-04 23:22:48.850	debug	(18549) photovoltaic inverter 3 enabled
solarviewdatareader.0	2020-05-04 23:22:48.834	debug	(18549) photovoltaic inverter 2 enabled
solarviewdatareader.0	2020-05-04 23:22:48.761	debug	(18549) photovoltaic inverter 1 enabled
solarviewdatareader.0	2020-05-04 23:22:48.289	debug	(18549) d0 converter: true
solarviewdatareader.0	2020-05-04 23:22:48.289	info	(18549) start solarview 10.0.40.46:15000 - polling interval: 1 Min. (06:00 to 23:00)
solarviewdatareader.0	2020-05-04 23:22:48.260	info	(18549) starting. Version 0.2.0 in /opt/iobroker/node_modules/iobroker.solarviewdatareader, node: v12.16.3, js-controller: 3.0.20
solarviewdatareader.0	2020-05-04 23:22:47.258	debug	(18549) statesDB connected
solarviewdatareader.0	2020-05-04 23:22:47.257	debug	(18549) States connected to redis: 0.0.0.0:9000
solarviewdatareader.0	2020-05-04 23:22:47.151	debug	(18549) States create System PubSub Client
solarviewdatareader.0	2020-05-04 23:22:47.149	debug	(18549) States create User PubSub Client
solarviewdatareader.0	2020-05-04 23:22:47.123	debug	(18549) Redis States: Use Redis connection: 0.0.0.0:9000
solarviewdatareader.0	2020-05-04 23:22:47.121	debug	(18549) objectDB connected
solarviewdatareader.0	2020-05-04 23:22:47.115	debug	(18549) Objects connected to redis: 0.0.0.0:9001
solarviewdatareader.0	2020-05-04 23:22:47.102	debug	(18549) Objects client initialize lua scripts
solarviewdatareader.0	2020-05-04 23:22:47.101	debug	(18549) Objects create User PubSub Client
solarviewdatareader.0	2020-05-04 23:22:47.100	debug	(18549) Objects create System PubSub Client
solarviewdatareader.0	2020-05-04 23:22:47.096	debug	(18549) Objects client ready ... initialize now
solarviewdatareader.0	2020-05-04 23:22:47.060	debug	(18549) Redis Objects: Use Redis connection: 0.0.0.0:9001
solarviewdatareader.0	2020-05-04 23:22:43.356	info	(2215) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
solarviewdatareader.0	2020-05-04 23:22:43.355	info	(2215) terminating
solarviewdatareader.0	2020-05-04 23:22:43.355	info	(2215) cleaned everything up...
solarviewdatareader.0	2020-05-04 23:22:43.354	info	(2215) Got terminate signal TERMINATE_YOURSELF

afuerhoff, 2020-05-05:
Wenn die Werte, die im Log angezeigt werden schon doppelt sind, dann kann der Adapter da nichts machen. Das sind die Rohdaten aus Solarview. Evtl. Solarview Konfiguration prüfen. 
Passen denn die Werte aus dem Log zu den in den Objekten gespeicherten Werte?

afuerhoff, 2020-05-07:
Hallo, konntest Du noch etwas herausfinden? Mir würde es helfen, wenn ich die Werte aus der Weboberfläche, die Werte aus dem Log und die Werte der Objekte als Bilder bekommen könnte. Dann könnte ich evtl. Rückschlüsse ziehen, wo der Fehler liegt. Evtl. könnte ich mal Kontakt mit dem Entwickler von Solarview aufnehmen.

vw-arneth, 2020-05-08:
Hallo, die webseite ist www.arneth-christian.de

08.05.2020 | 06:08:36.636 | solarviewdatareader. | 0 | (2237) 01*:   {01,08,05,2020,06,08,0000.0,00145,001664,00041805,00032,256,000.1,230,000.1,000,000.0,000,000.0,231,000.2,000,000.0,000,000.0,41},G
-- | -- | -- | -- | --
08.05.2020 | 06:08:36.640 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:08:39.635 | solarviewdatareader. | 0 | (2237) 02*:   {02,08,05,2020,06,08,0000.0,00151,001763,00044050,00034,254,000.1,254,000.1,000,000.0,000,000.0,230,000.2,000,000.0,000,000.0,40},D
08.05.2020 | 06:08:39.640 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:08:42.636 | solarviewdatareader. | 0 | (2237) 03*:   {03,08,05,2020,06,08,0000.0,00154,001707,00041577,00033,254,000.1,253,000.1,000,000.0,000,000.0,230,000.2,000,000.0,000,000.0,40},O
08.05.2020 | 06:08:42.640 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:08:45.640 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:09:24.635 | solarviewdatareader. | 0 | (2237) 00*:   {00,08,05,2020,06,09,0000.0,00450,005134,00127432,00105,000,000.0,000,000.0,000,000.0,000,000.0,000,000.0,000,000.0,000,000.0,00},#
08.05.2020 | 06:09:24.639 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:09:27.636 | solarviewdatareader. | 0 | (2237) 22*:   {22,08,05,2020,06,09,0005.9,00080,001985,00039420,00000,000,000.0,000,000.0,000,000.0,000,000.0,00},5
08.05.2020 | 06:09:27.639 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:09:30.635 | solarviewdatareader. | 0 | (2237) 10*:   {10,08,05,2020,06,09,0002.0,00092,001818,00015590,00148,000,000.0,000,000.0,000,000.0,000,000.0,00},3
08.05.2020 | 06:09:30.639 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:09:33.635 | solarviewdatareader. | 0 | (2237) 21*:   {21,08,05,2020,06,09,0000.0,00350,003905,00100650,00000,000,000.0,000,000.0,000,000.0,000,000.0,00},
08.05.2020 | 06:09:33.639 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:09:36.637 | solarviewdatareader. | 0 | (2237) 01*:   {01,08,05,2020,06,09,0000.0,00145,001664,00041805,00035,256,000.1,230,000.1,000,000.0,000,000.0,230,000.2,000,000.0,000,000.0,41},J
08.05.2020 | 06:09:36.642 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:09:39.637 | solarviewdatareader. | 0 | (2237) 02*:   {02,08,05,2020,06,09,0000.0,00151,001763,00044050,00037,254,000.1,254,000.1,000,000.0,000,000.0,230,000.2,000,000.0,000,000.0,40},H
08.05.2020 | 06:09:39.642 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:09:42.636 | solarviewdatareader. | 0 | (2237) 03*:   {03,08,05,2020,06,09,0000.0,00154,001707,00041577,00035,254,000.1,254,000.1,000,000.0,000,000.0,230,000.2,000,000.0,000,000.0,41},T
08.05.2020 | 06:09:42.641 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:09:45.640 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:10:24.636 | solarviewdatareader. | 0 | (2237) 00*:   {00,08,05,2020,06,10,0000.0,00450,005134,00127432,00151,000,000.0,000,000.0,000,000.0,000,000.0,000,000.0,000,000.0,000,000.0,00},
08.05.2020 | 06:10:24.641 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:10:27.637 | solarviewdatareader. | 0 | (2237) 22*:   {22,08,05,2020,06,10,0005.9,00080,001985,00039420,00000,000,000.0,000,000.0,000,000.0,000,000.0,00},-
08.05.2020 | 06:10:27.642 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:10:30.637 | solarviewdatareader. | 0 | (2237) 10*:   {10,08,05,2020,06,10,0002.0,00092,001818,00015590,00139,000,000.0,000,000.0,000,000.0,000,000.0,00},+
08.05.2020 | 06:10:30.645 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:10:33.636 | solarviewdatareader. | 0 | (2237) 21*:   {21,08,05,2020,06,10,0000.0,00350,003905,00100650,00000,000,000.0,000,000.0,000,000.0,000,000.0,00},
08.05.2020 | 06:10:33.640 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:10:36.637 | solarviewdatareader. | 0 | (2237) 01*:   {01,08,05,2020,06,10,0000.0,00145,001664,00041805,00050,256,000.1,230,000.1,000,000.0,000,000.0,230,000.2,000,000.0,000,000.0,41},?
08.05.2020 | 06:10:36.644 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:10:39.637 | solarviewdatareader. | 0 | (2237) 02*:   {02,08,05,2020,06,10,0000.0,00151,001763,00044050,00053,254,000.1,254,000.1,000,000.0,000,000.0,230,000.2,000,000.0,000,000.0,40},>
08.05.2020 | 06:10:39.641 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:10:42.637 | solarviewdatareader. | 0 | (2237) 03*:   {03,08,05,2020,06,10,0000.0,00154,001707,00041577,00052,254,000.1,254,000.1,000,000.0,000,000.0,230,000.2,000,000.0,000,000.0,41},K
08.05.2020 | 06:10:42.643 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:10:45.641 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:11:24.637 | solarviewdatareader. | 0 | (2237) 00*:   {00,08,05,2020,06,11,0000.0,00450,005134,00127432,00159,000,000.0,000,000.0,000,000.0,000,000.0,000,000.0,000,000.0,000,000.0,00},%
08.05.2020 | 06:11:24.641 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 06:11:27.637 | solarviewdatareader. | 0 | (2237) 22*:   {22,08,05,2020,06,11,0005.9,00080,001985,00039420,00000,000,000.0,000,000.0,000,000.0,000,000.0,00},.
08.05.2020 | 06:11:27.644 | solarviewdatareader. | 0 | (2237) connection closed
[... cut ...]
08.05.2020 | 07:08:36.680 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 07:08:39.674 | solarviewdatareader. | 0 | (2237) 02*:   {02,08,05,2020,07,08,0000.2,00152,001764,00044050,00356,300,000.6,297,000.6,000,000.0,000,000.0,229,001.6,000,000.0,000,000.0,42},h
08.05.2020 | 07:08:39.679 | solarviewdatareader. | 0 | (2237) connection closed
08.05.2020 | 07:08:42.675 | solarviewdatareader. | 0 | (2237) 03*:   {03,08,05,2020,07,08,0000.2,00154,001708,00041577,00353,303,000.6,301,000.6,000,000.0,000,000.0,228,001.6,000,000.0,000,000.0,42},e

afuerhoff, 2020-05-09:
Ich brauche auch noch die Objekte. Sonst kann ich das nicht miteinander vergleichen und auswerten. Möglichst vom gleichen Zeitpunkt, damit die Daten zusammenpassen. Deshalb hatte ich auch nach Bildern gefragt. Auf der Webseite sind die Daten ja immer aktuell und passen dann nicht mehr zum Log oder zu den Datenpunkten.

afuerhoff, 2020-05-17:
Hallo, ich habe Manfred Richter den Solarview Entwickler angeschrieben. Ich hoffe, dass er eine Lösung für das Problem hat. Ansonsten müsste ich noch als Workaround einen Umrechnungsfaktor mit einbauen. Die Rohdaten werden auf jeden Fall falsch von Solarview geliefert.

afuerhoff, 2020-06-02:
Hallo, kannst du mal den S0-Zähler 2 abfragen. Ich glaube dass 1 die Gesamtsumme aller angeschlossenen Zähler ausgibt.

afuerhoff, 2020-07-22:
Hallo, da ich keine Rückmeldung bekommen habe schließe ich den Issue.

closed 2020-07-22 by afuerhoff