afuerhoff/ioBroker.solarviewdatareader #11: Fehler im log

vw-arneth, 2020-04-09:
Hallo hab immer dies in meinem log stehen.

solarviewdatareader.0 | 2020-04-09 14:23:14.246 | warn | (2073) connect: checksum error
-- | -- | -- | --

afuerhoff, 2020-04-20:
Die Warnung habe ich auch. Kann ignoriert werden. Ich schaue mal bei Gelegenheit, ob meine Checksummen Routine noch einen kleinen Bug hat. In dem Moment werden die Werte nicht aktualisiert. Problem ist es nur, wenn jede Minute diese Meldung kommt.

afuerhoff, 2020-05-02:
Hallo, ich habe eine neue Testversion auf Github hochgeladen. Die Warnung sollte jetzt nicht mehr vorkommen. Falls der Fehler bei Dir nicht mehr kommt, würde ich die Version offiziell hochladen.

vw-arneth, 2020-05-03:
Hallo folgende fehler kommen im log....

solarviewdatareader.0 | 2020-05-03 21:05:06.838 | error | at TCP.onStreamRead (internal/stream_base_commons.js:186:23)
-- | -- | -- | --
solarviewdatareader.0 | 2020-05-03 21:05:06.838 | error | at Socket.Readable.push (_stream_readable.js:209:10)
solarviewdatareader.0 | 2020-05-03 21:05:06.838 | error | at readableAddChunk (_stream_readable.js:268:9)
solarviewdatareader.0 | 2020-05-03 21:05:06.838 | error | at addChunk (_stream_readable.js:286:12)
solarviewdatareader.0 | 2020-05-03 21:05:06.838 | error | at Socket.emit (events.js:310:20)
solarviewdatareader.0 | 2020-05-03 21:05:06.838 | error | at Socket.data (/opt/iobroker/node_modules/node-netcat/lib/client.js:38:30)
solarviewdatareader.0 | 2020-05-03 21:05:06.838 | error | at Client.emit (events.js:310:20)
solarviewdatareader.0 | 2020-05-03 21:05:06.838 | error | at Client.<anonymous> (/opt/iobroker/node_modules/iobroker.solarviewdatareader/main.js:419:88)
solarviewdatareader.0 | 2020-05-03 21:05:06.838 | error | (7455) ReferenceError: buf is not defined
solarviewdatareader.0 | 2020-05-03 21:05:06.831 | error | (7455) uncaught exception: buf is not defined

afuerhoff, 2020-05-03:
Hallo  kannst du nochmal updaten. Ich hatte zwischenzeitlich noch mal aktualisiert.

afuerhoff, 2020-05-05:
Ist der Fehler inzwischen behoben?

afuerhoff, 2020-05-07:
Fehler in 0.2.0 behoben

closed 2020-05-07 by afuerhoff