afuerhoff/ioBroker.solarviewdatareader #12: Keine WR Abfrage

mhelminger, 2020-12-06:
Hallo,

es werden von SolarView nur der EV sowie der D0-Zähler ausgelesen, die Wechselrichter
werden nicht ausgelesen - obwohl ausgewählt.

Anbei ein Debug-Log von ioBroker:

solarviewdatareader.0	2020-12-06 20:12:40.034	debug	(24595) connection closed
solarviewdatareader.0	2020-12-06 20:12:40.028	debug	(24595) 11*: {11,06,12,2020,20,12,0001.5,00001,002384,00010133,00473,000,000.0,000,000.0,000,000.0,000,000.0,00},
solarviewdatareader.0	2020-12-06 20:12:37.039	debug	(24595) connection closed
solarviewdatareader.0	2020-12-06 20:12:37.028	debug	(24595) 22*: {22,06,12,2020,20,12,0007.9,00149,005061,00044554,-0950,000,000.0,000,000.0,000,000.0,000,000.0,00},3
solarviewdatareader.0	2020-12-06 20:11:46.035	debug	(24595) connection closed
solarviewdatareader.0	2020-12-06 20:11:46.029	debug	(24595) 13*: {13,06,12,2020,20,11,0002.8,00003,002000,00024308,00275,000,000.0,000,000.0,000,000.0,000,000.0,00},
solarviewdatareader.0	2020-12-06 20:11:43.039	debug	(24595) connection closed
solarviewdatareader.0	2020-12-06 20:11:43.028	debug	(24595) 12*: {12,06,12,2020,20,11,0000.8,00001,001447,00042337,00000,000,000.0,000,000.0,000,000.0,000,000.0,00},
solarviewdatareader.0	2020-12-06 20:11:40.032	debug	(24595) connection closed
solarviewdatareader.0	2020-12-06 20:11:40.028	debug	(24595) 11*: {11,06,12,2020,20,11,0001.5,00001,002384,00010133,00423,000,000.0,000,000.0,000,000.0,000,000.0,00},
solarviewdatareader.0	2020-12-06 20:11:37.041	debug	(24595) connection closed
solarviewdatareader.0	2020-12-06 20:11:37.029	debug	(24595) 22*: {22,06,12,2020,20,11,0007.8,00149,005061,00044554,-0897,000,000.0,000,000.0,000,000.0,000,000.0,00},;
solarviewdatareader.0	2020-12-06 20:10:46.039	debug	(24595) connection closed
solarviewdatareader.0	2020-12-06 20:10:46.029	debug	(24595) 13*: {13,06,12,2020,20,10,0002.8,00003,002000,00024308,00275,000,000.0,000,000.0,000,000.0,000,000.0,00},
solarviewdatareader.0	2020-12-06 20:10:43.039	debug	(24595) connection closed
solarviewdatareader.0	2020-12-06 20:10:43.029	debug	(24595) 12*: {12,06,12,2020,20,10,0000.8,00001,001447,00042337,00000,000,000.0,000,000.0,000,000.0,000,000.0,00},
solarviewdatareader.0	2020-12-06 20:10:40.736	debug	(24595) system.adapter.admin.0: logging true
solarviewdatareader.0	2020-12-06 19:10:40.040	debug	(24595) connection closed
solarviewdatareader.0	2020-12-06 19:10:40.028	debug	(24595) 11*: {11,06,12,2020,20,10,0001.5,00001,002384,00010133,00418,000,000.0,000,000.0,000,000.0,000,000.0,00},
solarviewdatareader.0	2020-12-06 19:10:37.061	debug	(24595) connection closed
solarviewdatareader.0	2020-12-06 19:10:37.046	debug	(24595) 22*: {22,06,12,2020,20,10,0007.8,00149,005061,00044554,-1008,000,000.0,000,000.0,000,000.0,000,000.0,00},+
solarviewdatareader.0	2020-12-06 19:10:34.016	debug	(24595) self consumption meter 3 enabled
solarviewdatareader.0	2020-12-06 19:10:34.010	debug	(24595) self consumption meter 2 enabled
solarviewdatareader.0	2020-12-06 19:10:34.004	debug	(24595) self consumption meter 1 enabled
solarviewdatareader.0	2020-12-06 19:10:33.984	debug	(24595) photovoltaic inverter 4 enabled
solarviewdatareader.0	2020-12-06 19:10:33.964	debug	(24595) photovoltaic inverter 3 enabled
solarviewdatareader.0	2020-12-06 19:10:33.943	debug	(24595) photovoltaic inverter 2 enabled
solarviewdatareader.0	2020-12-06 19:10:33.922	debug	(24595) photovoltaic inverter 1 enabled
solarviewdatareader.0	2020-12-06 19:10:33.806	debug	(24595) d0 converter: true
solarviewdatareader.0	2020-12-06 19:10:33.805	info	(24595) start solarview 192.168.0.233:15000 - polling interval: 1 Min. (00:00:00 to 00:00:00)
solarviewdatareader.0	2020-12-06 19:10:33.776	info	(24595) starting. Version 0.2.1 in /opt/iobroker/node_modules/iobroker.solarviewdatareader, node: v10.23.0, js-controller: 3.1.6

Für Tipps wäre ich dankbar - ich habe schon alles mögliche probiert. Vielen Dank schon im Voraus.

afuerhoff, 2020-12-07:
Kannst du den Zeitraum mal so "00:00 bis 23:59" eingeben.

mhelminger, 2020-12-07:
Vielen Dank für die schnelle Hilfe. Es funktioniert perfekt. Danke.

afuerhoff, 2021-05-01:
Mit der neuen Vesion 1.0.0 wird der zweite Wert automatisch auf 23:59 geändert, wenn er auf 00:00 steht.

closed 2021-05-01 by afuerhoff